rmcanany/SolidEdgeHousekeeper #194: Add ability to update Material of part, but keep the face style the same

KGeetings, 2025-01-30:
We often have parts that we "color" with the material face style so that we know what to paint it. It would be amazing if we could update the material data, except keep the non-updated face style that we override.

rmcanany, 2025-01-30:
Which command are you using to update the material?  The two I can think of, `Update material from material table` and `Edit properties`, both have the option `Remove face style overrides`.  If that option is disabled, it is supposed to do what you want.

KGeetings, 2025-01-30:
I tried using the `Update material from material table` with `Remove face style overrides` unchecked, but the face style color was changed back to the default (black). I do also want to be clear, I am not using the "part painter" in Solid Edge to change the face colors, I'm changing the face style in the material table settings here (this blue face style got reverted to black, with remove face style overrides unchecked"):

[image]

KGeetings, 2025-01-30:
Is it possible that even though this setting was unchecked, that because my material bend radius was out of date, that it just updated everything anyway?

KGeetings, 2025-01-30:
Interestingly, when I attempted to use `edit properties` with `Remove face style overrides` unchecked then the face style stays the same. So it appears that maybe only the one command `Update material from material table` does not work?

KGeetings, 2025-01-30:
What's even more interesting, now that I look at the code between these two tasks, it seems as if they should both be calling the same exact `UpdateMaterialFromMaterialTable()`. So not entirely sure why one would remove the face style and the other wouldn't.

rmcanany, 2025-01-30:
Man, you got me.  I'll look at it tomorrow.

rmcanany, 2025-01-31:
I think I see what's going wrong.  I'm checking body, feature, and face FaceStyles, but not the MechanicalModeling.FaceStyle property.  The latter is the one that changes if the override comes from the Material Table like you describe.  I need to see if there's a way to deal with it using the Material Table API.  Changing the property name only is not enough.

Regarding it working with Edit Properties, the material update functionality only kicks in if %{System.Material} is changed.  Is it possible that your test did not change material?

rmcanany, 2025-02-03:
@KGeetings 

I may have it working.  There's a new option, `Update face styles`, which you would disable for your use case.  If you get a chance to try it, please let me know how it goes.

[image]

KGeetings, 2025-02-03:
Trying this crashed my Solid Edge and produced the following error:

```
Error processing file
        System.Exception: An unhandled exception has occurred. See inner exception for details. ---> System.Reflection.TargetException: Object does not match target type.
   at System.RuntimeType.InvokeDispMethod(String name, BindingFlags invokeAttr, Object target, Object[] args, Boolean[] byrefModifiers, Int32 culture, String[] namedParameters)
   at System.RuntimeType.InvokeMember(String name, BindingFlags bindingFlags, Binder binder, Object target, Object[] providedArgs, ParameterModifier[] modifiers, CultureInfo culture, String[] namedParams)
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.SolidEdgeDocument.get_Type()
   at Housekeeper.UtilsCommon.GetDocType(SolidEdgeDocument SEDoc) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsCommon.vb:line 160
   at Housekeeper.TaskUpdateFlatPattern.ProcessInternal(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateFlatPattern.vb:line 70
   at Housekeeper.HCIsolatedTaskProxy._Closure$__12-0`4._Lambda$__0() in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 196
   --- End of inner exception stack trace ---
   at Housekeeper.HCIsolatedTaskProxy.InvokeSTAThread[TArg1,TArg2,TArg3,TResult](Func`4 target, TArg1 arg1, TArg2 arg2, TArg3 arg3) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 207
   at Housekeeper.TaskUpdateFlatPattern.Process(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateFlatPattern.vb:line 37
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 497
```

KGeetings, 2025-02-03:
> I think I see what's going wrong. I'm checking body, feature, and face FaceStyles, but not the MechanicalModeling.FaceStyle property. The latter is the one that changes if the override comes from [...]

Regarding your question here. Yes, that is exactly what happened there. My test did not change the material, only the thickness, bend radius, and face style.

rmcanany, 2025-02-03:
I just pushed a new commit -- Solid Edge was taking a long time to Quit().  I added a bunch of .DoIdle commands where I am interacting with the material table.  

Not sure it's related, but could you please give it a try?

KGeetings, 2025-02-03:
Still getting the exact same error as above, even with the new commit.

rmcanany, 2025-02-03:
OK.  I think somehow the Application is getting hosed in TaskUpdateMaterialFromMaterialTable.  If you run TaskUpdateFlatPattern alone, do you still the problem?

KGeetings, 2025-02-03:
This issue does not happen when running TaskUpdateFlatPattern alone, everything looks good when doing just that task. Interestingly, after running just that task, and then only running the "TaskUpdateMaterialFromMaterialTable" task, I now get a different error

```
116.psm in C:\Users\Kenyon\Desktop\Test\SE Files v2
    Error processing file
        System.Runtime.InteropServices.COMException (0x800706BA): The RPC server is unavailable. (Exception from HRESULT: 0x800706BA)
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.SolidEdgeDocument.Close(Object SaveChanges, Object Filename, Object RouteWorkbook)
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 523

```

rmcanany, 2025-02-03:
I'm pretty sure the RPC server is the Solid Edge Application.  If you have time, maybe you could look over UtilsMaterials, lines 172-199.  You might spot something I'm doing wrong.

farfilli, 2025-02-04:
I have a working function to change material, will share tomorrow cause today I'm out of office

rmcanany, 2025-02-08:
I disabled that functionality for now.  I'm probably doing something wrong.  I sent a pared down program with just that code to my VAR.  Hopefully they'll figure something out.

rmcanany, 2025-02-08:
@KGeetings 

I may have stumbled on a way to make this work.  I added `SEDoc.Save()` after `SetMaterialPropValueToLibrary()` and `ApplyMaterialToDoc()`.  The changes were committed just now. 

To recap, here's what the code is doing
- Grab the face style from the part's material and the library's material.  
- Temporarily set the library's face style to that of the part.
- Check to see if the material is out of date, update if so
- Restore the library's face style

Below are a couple of things to try.  In a quick test, it worked for me.  I also added a second command like in your test and that worked, too.  (No guarantees it will work for you, of course.)

1
- In the part file, override the material face style
- Apply to part, but do not save the library
- Run Update material from material table with `Update face styles` and `Remove overrides` disabled
- The material should not be out of date, so should skip `ApplyMaterialToDoc()`
- Verify the part has the correct face style and the material table highlights it as overridden

2
- Same as above, except also override some other material parameter.  (I set elongation to 0.1)
- Again, apply to part, but do not save the library
- This time it should be out of date, so will execute `ApplyMaterialToDoc()`
- Verify part face style and material table override

3+
- Anything you think of

KGeetings, 2025-02-10:
I 100% should have brought this up earlier, but we are still using Solid Edge 2021 for most of our projects and thus, that is what I have been testing with. Since I was still getting an error on test #1, I started up a new Virtual Machine running the latest Solid Edge 2025 just to make sure it wasn't a Solid Edge version issue, and attempted the following tests, but as you can see, still receiving the same errors for both tests (same error occurred using both 2021 and 2025 versions):
```116.psm in C:\Users\Test Env\Desktop\SE Files v2
    Error processing file
        System.Exception: An unhandled exception has occurred. See inner exception for details. ---> System.Runtime.InteropServices.COMException: Unspecified error (Exception from HRESULT: 0x80004005 (E_FAIL))
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.MatTable.GetMaterialListFromLibrary(String bstrLibraryName, Int32& plNumMaterials, Object& listOfMaterials)
   at Housekeeper.UtilsMaterials.UpdateMaterialFromMaterialTable(Application _SEApp, SolidEdgeDocument _SEDoc, String _ActiveMaterialLibrary, Boolean _UpdateFaceStyles, Boolean _RemoveFaceStyleOverrides) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsMaterials.vb:line 176
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.ProcessInternal(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 175
   at Housekeeper.HCIsolatedTaskProxy._Closure$__12-0`4._Lambda$__0() in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 196
   --- End of inner exception stack trace ---
   at Housekeeper.HCIsolatedTaskProxy.InvokeSTAThread[TArg1,TArg2,TArg3,TResult](Func`4 target, TArg1 arg1, TArg2 arg2, TArg3 arg3) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 207
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.Process(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 134
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 497

```

I suppose I'm not entirely sure if I'm doing something wrong or not. But it's interesting that it seems to work for you, but not for me.

KGeetings, 2025-02-10:
I 100% should have brought this up earlier, but we are still using Solid Edge 2021 for most of our projects and thus, that is what I have been testing with. Since I was still getting an error on test #1, I started up a new Virtual Machine running the latest Solid Edge 2025 just to make sure it wasn't a Solid Edge version issue, and attempted the following tests, but as you can see, still receiving the same errors for both tests (same error occurred using both 2021 and 2025 versions):
```116.psm in C:\Users\Test Env\Desktop\SE Files v2
    Error processing file
        System.Exception: An unhandled exception has occurred. See inner exception for details. ---> System.Runtime.InteropServices.COMException: Unspecified error (Exception from HRESULT: 0x80004005 (E_FAIL))
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.MatTable.GetMaterialListFromLibrary(String bstrLibraryName, Int32& plNumMaterials, Object& listOfMaterials)
   at Housekeeper.UtilsMaterials.UpdateMaterialFromMaterialTable(Application _SEApp, SolidEdgeDocument _SEDoc, String _ActiveMaterialLibrary, Boolean _UpdateFaceStyles, Boolean _RemoveFaceStyleOverrides) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsMaterials.vb:line 176
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.ProcessInternal(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 175
   at Housekeeper.HCIsolatedTaskProxy._Closure$__12-0`4._Lambda$__0() in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 196
   --- End of inner exception stack trace ---
   at Housekeeper.HCIsolatedTaskProxy.InvokeSTAThread[TArg1,TArg2,TArg3,TResult](Func`4 target, TArg1 arg1, TArg2 arg2, TArg3 arg3) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 207
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.Process(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 134
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 497

```

I suppose I'm not entirely sure if I'm doing something wrong or not. But it's interesting that it seems to work for you, but not for me.

I did a few other tests:
1. If I have `Update face styles` enabled and `Remove overrides` disabled, then I get no error, and the face style is is updated. 
2. If I have `Update face styles` disabled and `Remove overrides` enabled, then I get the same error as above.

rmcanany, 2025-02-10:
That looks like a different error to me.  Last time SEApp was getting hosed.  In that error, MatTable.GetMaterialListFromLibrary(ActiveMaterialLibrary, NumMaterials, MaterialList) is bombing out.  

In previous SE versions, to use that command I had to strip off the ActiveMaterialLibrary extension.  You could try uncommenting line 120 in Utils.Materials.  It might cause other MatTable calls to fail, though.  If you can, you could try to make 2 versions of that variable -- one with the extension and one without.  Then see which command likes which version.

rmcanany, 2025-02-10:
I just checked and it looks like most of the MatTable commands I'm using were added in SE ST7, so should be available in SE 2021.  Anyway, if they weren't, I don't think it would compile on your system.

KGeetings, 2025-02-10:
You're correct, I should have worded that better. I more just meant that the error that I listed was the same between the two versions of Solid Edge, not that it was the same as the previous error from the comments last week. 

Anyway, I just tried uncommenting line 120 in Utils.Materials. In SE2021, I got the following error with `Update face styles` and `Remove face style overrides` both disabled:
```
116.psm in C:\Users\Kenyon\Desktop\Test\SE Files v2
    Error processing file
        System.Exception: An unhandled exception has occurred. See inner exception for details. ---> System.Runtime.InteropServices.COMException: Unspecified error (Exception from HRESULT: 0x80004005 (E_FAIL))
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.MatTable.GetMaterialListFromLibrary(String bstrLibraryName, Int32& plNumMaterials, Object& listOfMaterials)
   at Housekeeper.UtilsMaterials.UpdateMaterialFromMaterialTable(Application _SEApp, SolidEdgeDocument _SEDoc, String _ActiveMaterialLibrary, Boolean _UpdateFaceStyles, Boolean _RemoveFaceStyleOverrides) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsMaterials.vb:line 176
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.ProcessInternal(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 175
   at Housekeeper.HCIsolatedTaskProxy._Closure$__12-0`4._Lambda$__0() in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 196
   --- End of inner exception stack trace ---
   at Housekeeper.HCIsolatedTaskProxy.InvokeSTAThread[TArg1,TArg2,TArg3,TResult](Func`4 target, TArg1 arg1, TArg2 arg2, TArg3 arg3) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 207
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.Process(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 134
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Kenyon\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 497

```

And in SE2025
```
116.psm in C:\Users\Test Env\Desktop\SE Files v2
    Error processing file
        System.Exception: An unhandled exception has occurred. See inner exception for details. ---> System.Runtime.InteropServices.COMException: The remote procedure call failed. (Exception from HRESULT: 0x800706BE)
   at System.RuntimeType.ForwardCallToInvokeMember(String memberName, BindingFlags flags, Object target, Int32[] aWrapperTypes, MessageData& msgData)
   at SolidEdgeFramework.MatTable.SetMaterialPropValueToLibrary(String bstrMatName, String bstrLibraryName, MatTablePropIndex lPropIndex, Object varPropValue)
   at Housekeeper.UtilsMaterials.UpdateMaterialFromMaterialTable(Application _SEApp, SolidEdgeDocument _SEDoc, String _ActiveMaterialLibrary, Boolean _UpdateFaceStyles, Boolean _RemoveFaceStyleOverrides) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsMaterials.vb:line 235
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.ProcessInternal(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 175
   at Housekeeper.HCIsolatedTaskProxy._Closure$__12-0`4._Lambda$__0() in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 196
   --- End of inner exception stack trace ---
   at Housekeeper.HCIsolatedTaskProxy.InvokeSTAThread[TArg1,TArg2,TArg3,TResult](Func`4 target, TArg1 arg1, TArg2 arg2, TArg3 arg3) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\HCIsolatedTaskProxy.vb:line 207
   at Housekeeper.TaskUpdateMaterialFromMaterialTable.Process(SolidEdgeDocument SEDoc, Dictionary`2 Configuration, Application SEApp) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\TaskUpdateMaterialFromMaterialTable.vb:line 134
   at Housekeeper.UtilsExecute.ProcessFile(String Path, String Filetype) in C:\Users\Test Env\Documents\GitHub\SolidEdgeHousekeeper\My Project\UtilsExecute.vb:line 497

```

rmcanany, 2025-02-10:
Looking more closely at the code, the exception on line 235 is running the same command it already ran with success.  That was the problem I was trying to solve with the SEDoc.Save() insertions.  That is working here, but apparently not for you.  

I previously tested a *.par file.  Just tried a *.psm and that worked here, too.  I have no idea what's going on.

rmcanany, 2025-02-10:
One more thought.  SE 2021 had trouble on line 176.  You might try ActiveMaterialLibrary2 there and see what happens.

farfilli, 2025-02-11:
@rmcanany What about updating the material through Structured Storage? In this way you can easily skip the update of facestyle

rmcanany, 2025-02-11:
That's a great idea!  The best part is that it's already done.  I disabled it because of the face update issue.

That also tells me how to proceed in SE mode.  I don't need to use the material table methods for properties other than face style.  I can just update those directly.

Thank you!

rmcanany, 2025-02-16:
I played around with this a bit.  Hit a snag with Structured Storage -- density is a variable, not a property.  Couldn't see an easy way to access it, let alone change it.  Still waiting on my VAR for info on the MatTable exceptions.

I added some functionality to update face styles from a Finish property.  The user would need to create a face style with the same name as any finishes they use.  There's also a place where they can enter any finishes that don't change the material appearance.  Feedback welcome!

[image]

rmcanany, 2025-04-13:
Placing on hold until the material table PR is fixed.