rmcanany/SolidEdgeHousekeeper #195: Is it possible to also check sheet metal bend radius out of date?

KGeetings, 2025-01-30:
We occasionally get .psm files that have the correct sheet metal thickness, but the bend radius is out of date. Is there a way to grab that data from the Solid Edge SDK for us to check if the bend radius is out of date? I would love if this had the ability to just check if it's out of date and inform the user, instead of just updating it. The use case there is the occasional bend radius that needs to be custom, so changing it without the user's consent could cause issues. 

[image]

rmcanany, 2025-01-30:
So you would want to iterate over the bends in the file and see if each bend radius matches the one for that material in the material table?

If so, here's where to go looking for bend radius.

[image]

If you haven't played around with the material table API, TaskUpdateMaterialFromMaterialTable would be a place to start learning about it.

If you're interested in integrating into Housekeeper, that functionality might be at home in TaskCheckRelationships.  I'd probably need to change the name to something more general, but that's not a big deal.

KGeetings, 2025-01-30:
Yep, that's exactly the idea! Would want to check through each bend, see if it matches what we should expect from the material table, and just log if we see something out of place. I'll have to do some reading on the material table API.