rmcanany/SolidEdgeHousekeeper #200: File list updating wrong item with Sort = None and KeepDuplicates = True

rmcanany, 2025-02-11:
The reverse lookup table uses file name as the key.  Duplicate names cause the value to be overwritten.

Looks like `ListViewFiles.Items.Item(Filename)` returns the first occurrence of the filename in the list.

Might be able to use indices instead of names, but when processing selected files only, the index in `ListViewItems` and `FoundFiles` do not match.  Possibly adding a check that `FoundFiles` contains `ListViewItem` could resolve it.

Then, instead of this 
`FMain.ListViewFiles.Items.Item(FileToProcess).ImageKey = "Checked"`
Maybe this
`LVItemReverseLUT(idx).ImageKey = "Checked"`

farfilli, 2025-02-11:
Ok, got it.
I would in this case change the whole way to calculate the key, for example append the row number at the begin would be useful also to sort it correctly

rmcanany, 2025-02-11:
That sounds good.  I'll have a go at it later this week.

rmcanany, 2025-02-16:
Closing as complete.

closed 2025-02-16 by rmcanany