rmcanany/SolidEdgeHousekeeper #202: Collapsible sections in README.md

farfilli, 2025-02-19:
In my opinion, introducing collapsible sections in the README would help a user to browse it; at the moment you have to scroll a very lot

<details><summary><h1>Example A</h1></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

<details><summary><h2>Example B</h2></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>

<details><summary><h3>Example B.1</h3></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

<details><summary><h3>Example B.2</h3></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

<details><summary><h3>Example B.3</h3></summary>

<details><summary><h4>Example B.3.1</h4></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

<details><summary><h4>Example B.3.2</h4></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

<details><summary><h4>Example B.3.3</h4></summary>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
bla bla bla bla bla bla bla bla bla bla<br>
</details>

</details>

</details>

rmcanany, 2025-02-19:
I totally agree.  If I had an editor, maybe there would be less blah blah blah.

I actually started out in that direction, then stopped for some reason.  I will revisit.

farfilli, 2025-02-20:
Have a look at this: https://github.com/Mottie/GitHub-userscripts/wiki/GitHub-collapse-markdown

rmcanany, 2025-03-04:
Closing as complete