rmcanany/SolidEdgeHousekeeper #203: dft name match part name

tempod, 2025-02-24:
Hello.
First:
Congratulations again for the incredible work
Second:
In an old version there was a task to check that the dft file had the same name as the file it refers to.
It is not possible to do this with the new version?

Thank you

rmcanany, 2025-02-24:
There is a command, Part Number Does Not Match Filename.  Is that the one
you're thinking of?

If you need to check the whole name, you should now be able to do it with a
Property Filter.  Here's how I would set it up.  Note, the Value field uses
an index reference "|R1".  That is specifying the first attached model in
the drawing.  Also note, the formula is "NOT A".  I changed that using the
Edit Formula command on the toolbar.

[image: image.png]

tempod, 2025-02-25:
I can't get it to work
to test it i created a cube and two drafts. one with the correct name and one with the wrong name

[image]

these are the settings i used.

[image]

and this is the log

[image]

rmcanany, 2025-02-25:
[image]
I think that bug is fixed.  If you want to test it, email me and I'll reply with a link to my Google Drive.

rmcanany, 2025-03-04:
Hi @tempod , Just wondering if you got it to work.  Here's a picture of the setup I suggested.

[image]

tempod, 2025-03-06:
Hi.
Sorry for the incredible delay.
Under configuration --> templates --> customize
I had already inserted the variable "File Name (no extension)" but it didn't work.
The only solution to make it work is to put the configuration you posted above.
I just sent you an email. If you want I can test this function.

rmcanany, 2025-03-06:
Sent the link.  Please let me know how it goes.

tempod, 2025-03-13:
I finally did some testing in beta version.
Now the option also works when unchecking "run without solid edge"
[image]

But it seems to only check if the name is contained in the filename and not if it is exactly there.
If you have two pieces with a common string it will not flag it as an error
[image]

...and... one request
You can add this name control function in this space?
[image]

Thank you!

rmcanany, 2025-03-13:
I appreciate the feedback!

You're right, it checks if the part number is anywhere in the file name.  It works for us because our part number will be something like `7481-14102` while the file name will be like `7481-14102_bracket.psm`.  

You're not the first person to ask for something different.  I think it needs to be more general.  I could call the command "Check file name" instead.  Then I could provide a property formula and different comparison options.  For me, the formula would be `%{Custom.PartNumber}` and the comparison would be `contains`.  For you, the formula would be `%{System.File name (no extension)}` and the comparison would be `is_exactly`.  Would that work?

Not sure I understand the last item.  Are you asking for a filename check as an option to checking drawings?  If so, that sounds redundant to me.  I'm probably missing something; please elaborate.

farfilli, 2025-03-13:
I always thought that the field "Part number property" is where the part number is stored to be used in comparison with the filename.

What is the point of using %{System.File Name (no extension)|R1} in that field? Doesn't the reference file have a property to hold the part number? I can't believe the file name is the only place that holds that information.

rmcanany, 2025-03-13:
Yes, exactly.  Let me see what I can come up with.

tempod, 2025-03-18:
> Apprezzo il tuo feedback!
> 
> Hai ragione, controlla se il numero di parte è da qualche parte nel nome del file. Per noi funziona perché il nostro numero di parte sarà qualcosa come `7481-14102`mentre il nome del file sarà come `7481-14102_bracket.psm`.
> 
> Non sei la prima persona a chiedere qualcosa di diverso. Penso che debba essere più generale. Potrei chiamare il comando "Controlla nome file" invece. Quindi potrei fornire una formula di proprietà e diverse opzioni di confronto. Per me, la formula sarebbe `%{Custom.PartNumber}`e il confronto sarebbe `contains`. Per te, la formula sarebbe `%{System.File name (no extension)}`e il confronto sarebbe `is_exactly`. Funzionerebbe?

I think so

> Non sono sicuro di aver capito l'ultimo punto. Stai chiedendo un controllo del nome del file come opzione per il controllo dei disegni? Se è così, mi sembra ridondante. Probabilmente mi sfugge qualcosa; spiega meglio.

I was wondering if it was possible to add a check to automatically check that the dft name matches the psm, asm or par file name. Like to restore the old option "File name does not match the model file name"

> **Ah, aspetta, è solo un trucco per confrontare i nomi dei file. Sono d'accordo che le funzioni di confronto dovrebbero essere disponibili per quella funzione**

Exactly. Solid Edge automatically gives the dft the same name as the model. I think it's pretty standard. In our workflow we have very similar files. To create the dft of all the models we create one, copy it and with designa manager we change the link. That comparison function is useful to understand if we have replaced everything correctly.

rmcanany, 2025-03-18:
Got it.  It's on the list!

rmcanany, 2025-03-25:
@tempod Sent you an email.

-- Changed the command name to "Check Filename"
-- Changed input from a single property to a property formula
-- Added a new comparison option -- "Comparison is exactly"
-- Improved the [documentation](https://github.com/rmcanany/SolidEdgeHousekeeper/blob/ae00ef48cf33ac2b8e93420f11e8c8553051c3a0/HelpTopics.md#check-filename) (included an example for someone with your use case)

If you get a chance to test it, I would appreciate any feedback.

rmcanany, 2025-04-02:
Closing as complete

closed 2025-04-02 by rmcanany