rmcanany/SolidEdgeHousekeeper #210: House Keeper error "could not start solid edge..exiting"

DieCola85, 2025-05-13:
Hi everyone, my name's Diego and i'm a long time user of this beautiful app, which improves a lot my workflow. Sadly, since last Windows 11 update (dunno if related directly) i'm getting this error "could not start solid edge..exiting" no matter the task i select. I've tried to uninstall and reinstall House Keeper (2025.2, the lastest one) and no luck yet. I'm running Solid Edge 2024 on Windows 11. Any idea/advice? thank you in advance, have a nice day.

rmcanany, 2025-05-13:
Hi Diego,

I'm also running SE 2024 on Win11 and haven't seen that one.  If possible could you upload a zip of your Preferences directory (it's in the same directory where Housekeeeper.exe is stored.)

It's possible I messed up something with the latest version.  You could try downloading a previous version and try running that.

I'm assuming SE runs OK otherwise, right?  Oh, and if you haven't, it wouldn't be a bad idea to reboot and try again.

rmcanany, 2025-05-14:
I made a preview release for you to try.  It's out on my Google Drive; the link is below.  It won't fix the problem, but will hopefully provide a more informative error message.  Please take a screenshot of it and post back.

https://drive.google.com/file/d/1qK_J5utHIKMXPjDrkz0lGbfrYdFMLek_/view?usp=sharing

DieCola85, 2025-05-15:
hi @rmcanany thanks; as you said didn't work out but it says much more informations. New preferences attached

[Preferences.zip](https://github.com/user-attachments/files/20220437/Preferences.zip)

thanks again!

farfilli, 2025-05-15:
Please post a screenshot of the new error message

rmcanany, 2025-05-15:
Hi Diego, As Francesco says, we need to see a screenshot the reported error to figure out how to proceed.

DieCola85, 2025-05-16:
Hi, here you go, this is what i get. Thanks for the help.
[image]

DieCola85, 2025-05-16:
(it's italian, sorry for the language).

farfilli, 2025-05-16:
It may happen that the COM object got unregistered, it happened to me a couple of times especially when playing with multiple installations. In this case re-apply the latest SE update usually fixes the things.

DieCola85, 2025-05-20:
> It may happen that the COM objects got unregistered; it happened to me a couple of times, especially when playing with multiple installations. In this case,e re-apply the latest SE update usually fixes things.

Thanks, i've upgraded my Solid Edge with the lastest service pack, and it worked! :) thank you very much for the help, very appreciated.
Keep up the good work.
Have a nice day everyone.

(@farfilli i'm 3DieCola from 3Dcad.it forum)

rmcanany, 2025-05-20:
Thank you Francesco!!  I would have never figured that out.  Closing as completely conquered.

closed 2025-05-20 by rmcanany