rmcanany/SolidEdgeHousekeeper #215: Property not found errors handling

farfilli, 2025-06-17:
Today I wanted to see the %{System.Origination Date} and %{System.Last Saved} properties, in italian they are localized as "Data di origine" and "Data ultimo salvataggio". This date are available for sure but they are not correctly retrieved from SEHK.

Can you confirm is a localization problem and take care of it ?

[image]

I did try to use them to iteractively write them in Title property and found out that SEHK crashes if they are not found.

I provided a solution with #214 that is ok for manual usage, but not ideal for batch running; please provide your solution

rmcanany, 2025-06-18:
I might have this fixed.  In SSDoc, corrected names are used for the properties.  I wasn't properly handling `Origin Date`, 'Last Save Date' and a couple of others.  It worked for me with Edit Properties set like this.

[image]

Note, I think the date may be localized.  Not sure how picky it is.

farfilli, 2025-06-18:
@rmcanany those two properties are correctly working now and localized.
However, if in the FileListView I double click a property to edit and I input a not existing property name SEHK now back to crash, cause the property is not found and this case is not handled

You can do youself by double click a property in the FileListView, and insert a property with a type, example use **%{System.Titleeee}** and you will see the error I mean

rmcanany, 2025-06-18:
Where is it crashing?  I double-clicked a property column, entered %{System.Titleee} and hit Return.  On line 3272 of Form_Main, SSDoc.SetPropValue returned False, so it ignored the change and reverted to the previous value.

rmcanany, 2025-06-18:
I thought of a place where it might be going wrong.  Could you check your Configuration Tab Templates Page and try to Update your properties?  There is supposed to be a check for that, but it may not be working.

farfilli, 2025-06-19:
The file is writable, and writing a simple text works. Also, writing an existing property works.
The problem occurs when you try to insert a non-existent property into an existing property manually.

[image]

rmcanany, 2025-06-19:
Added a check for invalid entry in `editbox_LostFocus`.  Pushed a commit just now.

farfilli, 2025-06-19:
Just tested and it works as expected.

Out of curiosity, where does the localization of the properties occours?
In detail I'm interested to find the "Created date" to compare with "Origination date" and "Last save date"

rmcanany, 2025-06-19:
Good deal.

The localization happens when the user runs Update on the Templates page.  That launches `HCPropertiesData.Populate`, which scans all template property sets by index number.  A lookup table in `HCPropertiesData.New(_PropertySetActualName As String, _PropID As Integer)` provides the English name, type and a couple of other things; the templates are the source of the (sometimes) localized name. 

I'm not sure what the deal is with `Created date`.  It's not in my lookup table.  I had assumed it was another name for `Origination date`, but I'm not sure.

That's a good idea about adding some more info on interacting with the ListView.  I'll take a look at that.

I'm going to leave this issue open for now.  Please let me know if you run into anything else.

farfilli, 2025-06-20:
It seems that "CreationTime" is a property of [FileSystemInfo ](https://learn.microsoft.com/it-it/dotnet/api/system.io.filesysteminfo?view=net-8.0)and perhaps is not accessible through the StructuredStorage

closed 2025-07-03 by rmcanany