rmcanany/SolidEdgeHousekeeper #224: Remove from list

ih0nza, 2025-07-07:
Hello,
When I use right click on file then "Remove from list" file is moved to executed files. But when I want process this I need to "Update" and after update are that executed files is again loaded to the rest of files so it is impossible to use "Remove from list" feature.

rmcanany, 2025-07-07:
Odd.  I just tried it and it worked fine.  Excluding from list did not set the List Out of Date flag.

Could you tell me the steps you took?  Like how the files were added to the list (top level assy search, drag/drop, etc.)  Maybe include screen shots.  Also, are you using the preview version or the released V2025.2?

rmcanany, 2025-07-07:
I forgot that the Delete and Backspace keys did that.  I see what's happening.  I'll get it fixed.  

I don't see the same for the Remove from List method.  Might be some sort of edge case.

farfilli, 2025-07-07:
@rmcanany No, it's fully migrated to `ListViewSources `. If you found a reference to it, it's a mistake

ih0nza, 2025-07-08:
Hello, I apologize, next time I will specify the version I am using. I am currently testing 2025.3 beta. And the problem, as @farfilli wrote, is that the DEL key causes the issue—if I do it via the right-click option, it seems to work fine.

farfilli, 2025-09-02:
Perhaps it's time to reduce the needings to use the update button.

Or change the way the lists are handled, something like a list source is 'consumed' after the first update. It just became a list of file in the list view and the source removed

farfilli, 2025-09-03:
Or another approach would be to collect the list of excluded items and restore them to the excluded list after the update

rmcanany, 2025-09-03:
I like that.  Nice and clean, as usual.

I had another thought.  I was initially picturing Preset exclusion options on the Configuration Tab -- General page.  That panel is a mishmash of unrelated settings already.  Adding more was giving me pause.  

How about a separate options button on the Presets toolbar instead?  There we could, over time, add any number of related settings, including items to exclude.  Being focused on Presets, in my opinion, it wouldn't add to the cognitive load.