rmcanany/SolidEdgeHousekeeper #225: Move file to recycle bin

farfilli, 2025-07-07:
If your files reside on a network drive they are deleted permanently **without a warning.**
I accidentally deleted an hundred files for that reason.

Also I would add a button to just delete files from the list and not from the drive; being sure to clearly specify the differences

rmcanany, 2025-07-07:
Ugh, sorry about that.  Good ideas about the warning and deleting just from the list.  Let me see what I can do.

farfilli, 2025-07-08:
Network files deletion warning works.

I also checked the remove from list that works; however, the "exclude from processing" menu item doesn't work, and also the DEL key ( I would use it to exclude from processing, which is less intrusive compared to remove from list and move to recycle bin)

I don't have other fixes in mind at the moment, candidate for a new release but an interesting feature would be good; I open a new issue then you choose what to do about it

rmcanany, 2025-08-25:
Closing as complete.