rmcanany/SolidEdgeHousekeeper #231: Issue Adding Files

collands, 2025-08-22:
When using the "Add Single File" or "Add Files From List" buttons, files are not added correctly if a folder name (anywhere in the directory structure) exists that contains a comma. Adding a single folder or a folder with subfolders works properly. If a folder is added, and the list updated, then exported, that exported list still does not populate files correctly when imported.

(In my use case, the files are being accessed on a synced folder on a company SharePoint. Because Microsoft has our company name ending in ", Inc." I can't rename these folders. I had to work around this by mapping the local copy of the SharePoint folders to a drive letter. )

Steps to Reproduce:
- Create a folder with a comma in the name, such as "C:\Test, Folder\"
- Add any Solid Edge file to the new folder.
- Add that file to the File Sources using the "Add single file" button.
- Click Update:

Alternatively:
- Add several Solid Edge files to "C:\Test, Folder\"
- Add that folder using the "Add single folder" button.
- Click Update.
- Observe files in "C:\Test, Folder\" populate in File List.
- Export File List to text.
- Clear Sources and Files.
- Import previously exported files.
- Click Update.

Expected Behavior:
- File(s) appears in file list.

Actual Behaviour:
- No file appears.

rmcanany, 2025-08-22:
Appreciate the clarification.  I'll have a look, hopefully next week.

rmcanany, 2025-09-18:
Closing as complete.  Thank you for the bug report!