rmcanany/SolidEdgeHousekeeper #235: Update Part Copies DocType Task Update Part Copies not recognized

sven-11111, 2025-09-17:
This message comes when I select update Part coppies with update parent doxcuments.
I have to acknowledge this message to continue with the task.
Can I disable this message?

[image]
[image]

rmcanany, 2025-09-17:
That's very odd.  I'm not seeing that here.  Does it do that on all files?  

Are you using a published release, or a clone from GitHub?  If it's a release, which one?

sven-11111, 2025-09-17:
Yes on all files.
I use an published release 2025.2.

rmcanany, 2025-09-18:
OK, so disabling the message box is all you really need, right?

sven-11111, 2025-09-18:
Yes